dotMorten/NmeaParser #13: UWP projects in VS 2015 RTM

syabiku, 2015-10-18:
Thank you for this great library. I've been able to use it for Windows 8.1 store apps with no issues. I am now trying to use with UWP (Win 10) apps because I am eager to try the serial port device that is now available to us in UWP. However, the UWP project code is not compatible with Visual Studio 2015 RTM (you built them with a pre-release version of the Win 10 SDK), and I'm having trouble importing/converting them to VS 2015 RTM. Do you have any plans to update these projects?

dotMorten, 2015-10-18:
Good call! I'll try and get it done tonight.

dotMorten, 2015-10-19:
Done!

closed 2015-10-19 by dotMorten

syabiku, 2015-10-19:
Much appreciated! All working well, after an oddity on my end:

I am using a BU-353-S4 USB GPS receiver.  It uses a Prolific PL-2303 HXD USB to serial bridge, and I'm on the most recent (9/4/2015) drivers from Prolific.

Win32 apps recognized the NMEA stream coming off the COM port just fine, but my Windows 10/UWP app (and your test UWP app) would not read any incoming data from the serial port device. After comparing with a working UWP serial example (https://ms-iot.github.io/content/en-US/win10/samples/SerialSample.htm), I made the following change to your test app:
```
    serialDevice.BaudRate = 4800;
    var _foo = serialDevice.BaudRate;  // had to insert this
    serialDevice.DataBits = 8;
```
It's as if the the baud rate setting wouldn't stick until it was accessed. Weird!

Anyway, this is probably a bug in the Prolific device driver and/or my specific setup/hardware. I wanted to share this with you in case other people in your line of work are using USB devices in UWP apps with similar hardware.