dotMorten/NmeaParser #16: Cant import into xamarin

welshrat, 2016-01-14:
Hi there, just tried to add the package to a xamarin android solution and get the following message.

"You are trying to install this package into a project that targets 'MonoAndroid,Version=v4.4', but the package does not contain any assembly references or content files that are compatible with that framework."

dotMorten, 2016-01-14:
How are you adding the package? Nuget or via source? What version of xamarin? Could you please provide some repro steps?

welshrat, 2016-01-14:
Apologies, I was attempting to add via the package manager in xamarin, however I have downloaded the zip file and just this second compiled the android version. and was about to close the issue. I had just assumed I could import this way

dotMorten, 2016-01-14:
Sounds like the nuget package is broken. Let's keep this issue open to track that

dotMorten, 2017-03-07:
Can't reproduce. Package imports OK

closed 2017-03-07 by dotMorten