dotMorten/NmeaParser #20: PUBX strings

SteveK-GS, 2016-04-15:
Are you familiar with PUBX strings, are they within the scope of NmeaParser, and could support be added? I gather they come from GPS hardware provided by a company called u-blox.

I've been given a protocol to write a decoder for and PUBX is there amongst various NMEA types all of which are supported by the library except for the former.

dotMorten, 2016-04-15:
I wasn't aware until now. Looks like you can parse a PUBX just fine with this parser - the only worry I have is that the first value doesn't define the message type, but the first two values do. That's a little bit against how the API has been designed so far.

dotMorten, 2016-04-15:
Looking at the implementation we would have to change the reflection-based code that looks up the supported messages to also support a factory create approach for the messages that need further analysis. That's definitely doable, but I'm wonder if it's worth doing in a generic library, rather than just registering the "PUBX" message type from an extension and do additional parsing in the extension.

dotMorten, 2016-09-28:
Closing as out of scope

closed 2016-09-28 by dotMorten