dotMorten/NmeaParser #21: SerialPortDevice does not exist

fjclip99, 2016-06-05:
I am creating a visual studio universal app and i am using the UWP sample... But somehow on the line `var device = new NmeaParser.SerialPortDevice();` i get the error 

> SerialPortDevice does not exist

dotMorten, 2016-06-05:
Are you using the nuget package? Dont think I updated it with UWP support. You'll have to reference the uwp project instead

fjclip99, 2016-06-05:
I have the SharpGIS.NmeaParser v 1.5.1 installed. How do I reference the uwp project?

dotMorten, 2016-06-05:
Download the source, add project to your solution, then add a project reference.

fjclip99, 2016-06-05:
Ok, I included it and added the reference, but now i get the error under: `NmeaParser.SerialPortDevice` saying:` Predefined type 'System.Void' is not defined or imported`.

fjclip99, 2016-06-05:
Here is the code: http://prntscr.com/bcnf9j

fjclip99, 2016-06-05:
Ok, i figured it out - needed to uninstal the nuget package... 

So now i have another question:
On the event: 
`private void device_MessageReceived(object sender, NmeaParser.NmeaMessageReceivedEventArgs args)` How can I make that the received data would be saved in an string.
Also can I use the Pi's RX/TX to communicate?

fjclip99, 2016-06-05:
Ok, fixed that. Now I have one more question. Is the `private void device_MessageReceived(object sender, NmeaParser.NmeaMessageReceivedEventArgs args)` event called always there is incomming data on the serial port?

dotMorten, 2016-06-06:
It's called every time there's a valid full message received (so normally yes always)

fjclip99, 2016-06-06:
I am using just this code for now but i am not getting the event triggered:

```
private void device_MessageReceived(object sender, NmeaParser.NmeaMessageReceivedEventArgs args)
{
    textBlock.Text = "EVENT TRIGGERED";
}
```
And the init. code:
```
private async void InitSerial() // Initialize Serial
{
    string dev = SerialDevice.GetDeviceSelector("UART0");
    var inf = await DeviceInformation.FindAllAsync(dev);
    SerialPort = await SerialDevice.FromIdAsync(inf[0].Id);
    dataReader = new DataReader(SerialPort.InputStream);

    SerialPort.BaudRate = 9600;
    SerialPort.Parity = SerialParity.None;
    SerialPort.StopBits = SerialStopBitCount.One;
    SerialPort.DataBits = 8;
    SerialPort.Parity = SerialParity.None;
    SerialPort.Handshake = SerialHandshake.None;
    var device = new NmeaParser.SerialPortDevice(SerialPort);
    device.MessageReceived += device_MessageReceived;
    await device.OpenAsync();
}
```
Oh, and i am sending a string from A0 to A255.

dotMorten, 2016-06-06:
Did you try the test app and see if that work? Are you sure those are the correct parameters for your device?

dotMorten, 2016-06-06:
Updated the nuget package to 1.6.0 with support for UWP (and iOS and Android - Xamarin) https://www.nuget.org/packages/SharpGIS.NmeaParser/1.6.0

fjclip99, 2016-06-06:
The parameters are correct i think, because the SerialWrite that i created is working and it is sending, however the event doesen't want to trigger... I have installed the 1.6.0 and still nothing...

dotMorten, 2016-06-06:
How do you know you're sending? What GPS device is this where you can just send 0..255 and know it received it? Very few GPS devices support being written to and if they do, it's usually only RTCM data.

Serial ports are old tech. If you get any of these parameters wrong, nothing will happen. No error, just nothing. No data is sent or received.

dotMorten, 2016-06-06:
Also are you sure about your "UART0" parameter ?

fjclip99, 2016-06-06:
No no, i am not using the GPS device... I forgot to mention.... I am using an Arduino... I need to make a communication between the Arduino and the Raspberry PI... I have the SerialWrite function created, it is sending the data and the Arduino is receiving it... Now I only need to make the SerialRead on the Raspberry pi (Windows universal app) so that i can receive messages like (A255, B125, etc...).

dotMorten, 2016-06-06:
Huh? That's not what this library is for. You don't need this library at all.

closed 2016-06-06 by dotMorten