dotMorten/NmeaParser #23: How to connect to sensors not on serial ports?

pksorensen, 2016-08-28:
I am abit out of my domain here and sorry if the question do not make sense, as i might not know enough to ask the proper question.

I have a tablet with windows 10 and in the device manager, there is a sensors tab, which has a Broadcom GNSS 4752 Geolocation Sensor. Do these also have a com port that I can get data from or is there no way to use this library against such ?

dotMorten, 2016-08-28:
That's not a NMEA based device. Instead you can use the built-in location APIs to get a location, but you wouldn't be able to get some of the extra details the NMEA sentences provide.

closed 2016-08-28 by dotMorten

pksorensen, 2016-08-28:
I managed to use something called a GPSReverse tool that mapped it to a virtual com port, but it keept printing 0 for lat,lng sadly.

Thanks for clarifying