dotMorten/NmeaParser #34: NmeaDevice::CloseAsync does not close serial devices correctly (SerialPortDevice .CloseStreamAsync does not close passed stream )

mb12, 2017-07-19:
The current implementation of SerialPortDevice .CloseStreamAsync  does not Dispose the passed stream.  So if you try to open the device again after calling CloseAsync on it, it would fail.

So applications with the following kind of usage pattern will not work:

OpenDevice
....do something with NMEA data..
CloseDevice

**OpenDevice again          ============> This will fail because device is not closed correctly.**

```
void OpenDevice() {
............
...........
	m_device = new NmeaParser.SerialPortDevice(serialDevice);
	m_device.MessageReceived += device_NmeaMessageReceived;
        m_device.OpenAsync();
............
...........
}
void CloseDevice() {
        m_device.CloseAsync();
        m_device = null;
}
```

clangridge, 2017-09-06:
Also running into an issue with this as the application isn't closing because the it is still receiving events from the COM Port.  I've added code to the closing event of the form, but the Sub I have set up to handle events from the class keeps being called, blocking the closing from happening.

dotMorten, 2018-07-26:
I'm a little confused by this. The method calls Close, and the documentation for close clearly states the stream will be disposed too:

>   Closes the port connection, sets the System.IO.Ports.SerialPort.IsOpen property to false, and disposes of the internal System.IO.Stream object.

Is this .NET or UWP?

mb12, 2018-07-26:
This is how.

dotMorten, 2018-07-26:
Did you dispose the `SerialDevice` that you parsed in before creating a new one? Or did you reuse the same instance?

Also in your code above, I don't see any awaits on open and close. Are these calls completed before calling the next one?

clangridge, 2018-07-26:
Mine is .NET, WPF.  It is not related to #33.  I just called the library as is; I made no changes to.  The CloseAsync function says it has run successfully, but the class is still responding to data being received from the COM port, which tells me that the connection to the COM port has not actually closed.

dotMorten, 2018-07-28:
> the class is still responding to data being received from the COM port

I don't see how that's possible, as the thread reading the serial port is shut down ([the close task won't return until that thread has shut down](https://github.com/dotMorten/NmeaParser/blob/master/src/NmeaParser/NmeaDevice.cs#L137) and I've not been able to reproduce this behavior myself.
I wonder if there's more to it. Any chance you could provide a reproducer sample?

clangridge, 2018-08-02:
Are you sure that thread has shut down?  I ended up writing my own NMEA Parser class (https://github.com/clangridge/NMEAParser) and I had to wait for the serial com port to raise its Disposed event, even though the properties of the COM port would say it was stopped.

I don’t know what you mean by reproducer sample.  I’m using an actual, physical 9-pin COM port to receive the NMEA data, so any file I send you would not be reproducing the conditions that I am testing with.

Clark Langridge, B.Sc (physics), B.ScE (geomatics)
GIS User Analyst/Analyste SIG de l’utilisateur
Information Systems and Departmental Services Branch / Direction des systèmes d’information et des services ministériels
Departments of Energy and Resource Development and Agriculture, Aquaculture and Fisheries/Ministères du Développement de l’énergie et des ressources et agriculture, aquaculture et pêches
Office Phone: 506-453-7106

Whenever you find yourself on the side of the majority, it is time to pause and reflect.
-Mark Twain

dotMorten, 2018-08-02:
Thank you! I had not noticed the Disposed event on the SerialPort. That's very helpful!

dotMorten, 2018-08-02:
wrt to a reproducer sample, I just meant an app that opens/closes a serial port and causes the crash, and assume I also have a serial port and a device connected to it.

dotMorten, 2018-08-02:
Since the developer creates the serial port and passes it to the NMEA Parser, it's the developer's responsibility to dispose of the serial port, and ensure it's disposed before opening it up again.

All the NMEA Parser is doing is opening and closing the port. It would not be correct if the API would be disposing objects that it is not the creator of.

@mb12 Can you confirm that if you dispose the serial port and wait for the disposed event before opening up the port again, it works? (the sample app should probably be doing this again)

dotMorten, 2019-02-25:
Closing - feel free to open again if you have any extra detail to share or disagree with my previous comment

closed 2019-02-25 by dotMorten