dotMorten/NmeaParser #35: Does not parse NMEA strings from GLONASS and GPS devices

clangridge, 2017-09-06:
For some weird reason, the people who maintain the NMEA standards decided that for NMEA strings coming off a receiver that was GLONASS and GPS enabled, they would preface their strings with $GN rather than $GP for just GPS receivers [https://www.trimble.com/OEM_ReceiverHelp/V4.44/en/NMEA-0183messages_GNS.html](url) See field 0.

I've attached the output from a GLONASS + GPS receiver in a Panasonic FZ-G1 tablet.  Its output is pretty standard for other receivers that I have worked with that use both constellations.
[fzg1.txt](https://github.com/dotMorten/NmeaParser/files/1282254/fzg1.txt)

dotMorten, 2017-12-21:
I've added some glonass and galileo messages, plus several of the combined GN messages to v1.9.0. Let me know if it covers what you need.

clangridge, 2017-12-21:
The issue around serial COM ports not closing was the bigger one for us, so we just built our own class to do NMEA parsing.

Clark Langridge, B.Sc (physics), B.ScE (geomatics)
GIS User Analyst/Analyste SIG de l’utilisateur
Information Systems and Departmental Services Branch / Direction des systèmes d’information et des services ministériels
Departments of Energy and Resource Development and Agriculture, Aquaculture and Fisheries/Ministères du Développement de l’énergie et des ressources et agriculture, aquaculture et pêches
Office Phone: 506-453-7106

Whenever you find yourself on the side of the majority, it is time to pause and reflect.
-Mark Twain

closed 2018-04-02 by dotMorten