dotMorten/NmeaParser #36: NmeaDevice::ProcessNmea multipart message cache questions

terribletim, 2017-10-21:
With reference to NmeaDevice [ProcessNmea](https://github.com/dotMorten/NmeaParser/blob/master/src/NmeaParser.Shared/NmeaDevice.cs#L157)

1. Is a **Lock** object required when working with **MultiPartMessageCache**?
2. Should there be a **Return** after [Line 172](https://github.com/dotMorten/NmeaParser/blob/master/src/NmeaParser.Shared/NmeaDevice.cs#L172)? ie if something is out of order, clear cache and don't raise event
3. Likewise, should **IF** statement at [Line 182](https://github.com/dotMorten/NmeaParser/blob/master/src/NmeaParser.Shared/NmeaDevice.cs#L182), have an **ELSE** with a **Return** if the dictionary isn't full? (ie wait for all messages before raising event)

Thanks and regards
TT

terribletim, 2017-10-22:
Also, at [Line 185](https://github.com/dotMorten/NmeaParser/blob/master/src/NmeaParser.Shared/NmeaDevice.cs#L185) should 
`args.MessageParts = dic.Values.ToArray();`  
be changed to
`args.MessageParts = dic.Values.Cast<NmeaMessage>.ToArray;`

Regards

TT

dotMorten, 2019-02-25:
1. No shouldn't be, as the messages are not passed in a multi-threaded manner.
2. No this typically only happens in the beginning where you might start getting messages in-between.
3. The idea is to raise an event for each message - you'd have to check if all messages are there or not if you only care about the last one.

closed 2019-02-25 by dotMorten