dotMorten/NmeaParser #37: NmeaMessage.Parse() throws FormatException on some lines

EvAlex, 2018-01-09:
I use `NmeaMessage.Parse()` to parse lines in NMEA file. Almost all lines are parsed correctly, but not all of them. About 170 out of 1500 lines of my file fail to be parsed - `NmeaMessage.Parse()` throws FormatException originating from `double.Parse()` in `SatelliteVehicle` c'tor.

This issue is silently ignored in `NmeaDevice.ProcessMessage()` https://github.com/dotMorten/NmeaParser/blob/master/src/NmeaParser/NmeaDevice.cs#L145

Is this by design and am I mising something?

Example line reproducing the issue:
`NmeaMessage.Parse("$GPGSV,4,4,16,30,40,104,47,40,25,159,32,41,15,129,36,195,,,35*75")`

dotMorten, 2018-01-09:
The message looks invalid/incomplete. Here's the breakdown of the values:

- `4` = Total Messages
- `4` = Message number
- `16,30,40,104`: Prn=`16`, Elevation = `30`, Azimuth = `40`, SignalToNoise: `104`
- `47,40,25,159`: Prn=`47`, etc
- `32,41,15,129`: Prn=`32` etc
- `36,195,,`        : Incomplete Satellite
- `35`                : Incomplete Satellite
- `*75` - Checksum

From the spec, only the SNR value is optional.

dotMorten, 2018-01-09:
To answer your question about behavior:
- Calling Parse directly will throw, so you know what failed to process.
- The message processor does not throw - I don't want your entire app to come crashing down on you, just because there was a glitch in one of the messages (there wouldn't be a way to try/catch that since it's running in it's own thread)

EvAlex, 2018-01-10:
I see, thank you for explaination. The behavior is reasonable. 

Another question then. Is it possible to restore missing values based on existing ones and the checksum value? I implemented MRZ string restoration in some cases based on checksum and hope this is possible here too.

It's a pity I got the file with such incorrect/incomplete lines from a real car dvr. I wonder if this behavior is a common thing.

dotMorten, 2018-01-10:
The checksum is very basic and way too many permutations would create the same checksum, so no that's not possible. Also the checksum checks out. This is the string the device meant to send. I just have no idea why it's sending that.
There's plenty of devices out there being quite lenient on the spec. Sure I could fallback and just return zero or NaN values, but I think it's safer to assume the message is invalid and not parse it. And the example you sent, that doesn't even seem to be sufficient, as the last part of message doesn't even have a full satellite listed in it.

You might want to chat with the manufacturer and ask what they were thinking.

EvAlex, 2018-01-10:
I see, thank you! I guess the issue should be closed now.

closed 2018-01-10 by dotMorten