dotMorten/NmeaParser #39: Parser for handling raw strings?

asfarley, 2018-05-10:
I had a scenario where I'd like to parse strings (not files or streams) - of course I could just write my string to a temp file, but it'd be nice to have a built-in parser for handing this.

dotMorten, 2018-05-20:
You will need to wrap your strings in a stream. Check out the buffered stream device in the file stream class:
https://github.com/dotMorten/NmeaParser/blob/master/src/NmeaParser/BufferedStreamDevice.cs#L81

closed 2018-05-20 by dotMorten

davidce, 2024-08-05:
check  https://github.com/dotMorten/NmeaParser/blob/main/src/UnitTests/NmeaParser.Tests/BufferedStringDevice.cs and https://github.com/dotMorten/NmeaParser/blob/main/src/UnitTests/NmeaParser.Tests/DeviceTests.cs