dotMorten/NmeaParser #41: Cannot add custom Nmea message type

XPD-Kasun, 2018-07-24:
I tried to add a custom nmea type called GPVTG. 
 ```
[NmeaMessageType("GPVTG")]
  public class GPVTG : NmeaMessage
    {
        protected override void OnLoadMessage(string[] message)
        {
        }
    }
```

But this method (OnLoadMessage) doesnt get called even GPVTG message comes through the COM port nmea channel. Is there any registration needs to be done other than adding NmeaMessageType attribute.

dotMorten, 2018-07-24:
It would only work if it's part of the same assembly as the other assemblies.

XPD-Kasun, 2018-07-24:
I used the nuget version of this parser. So I cannot modify the assembly. In order to add this to the same assembly I have to rebuild from the source, dont I?

dotMorten, 2018-07-24:
Yes

dotMorten, 2018-07-24:
Yes. Alternatively, just listen for messages, and when you get an "UnknownMessage", just parse it yourself then and there.

XPD-Kasun, 2018-07-25:
I think it is the way to go.

closed 2018-07-25 by XPD-Kasun

dotMorten, 2018-07-26:
Also I added issue #42