dotMorten/NmeaParser #48: Generalized version for handling input / output from different sources?

hansmbakker, 2019-02-06:
Hi @dotMorten,

I was looking for a library to handle input / output from a serial port combined with incoming message parsing. I found this library which looked as a good start - this was the best I found besides building something based on [System.IO.Ports](https://www.nuget.org/packages/System.IO.Ports/).

The generalization could be that you can e.g. pass it your own parser implementation and set your own message separator. I could also imagine someone writing another interface (like a network interface implementation).

I uploaded a draft version at https://github.com/hansmbakker/InterfaceDevice (InterfaceDevice for lack of a better name), and was wondering whether you'd be interested in something like this? In any case I'd be interested in hearing your feedback - maybe you have ideas on this you want to share?

The changes I made were:
- carving out the NMEA-specific parts like the messages and the parser
- removing the multipart-functionality
- giving classes generic names
- allowing for a custom message separator
- allowing to specify an incoming message parser that implements a certain interface.

dotMorten, 2019-02-08:
@hansmbakker thanks for the detailed submission, and for discussing in an issue first.

However considering the purpose of this library is very specific to NMEA, I don't really see myself wanting to also support a much more generic parser on top, that to me seems to merely be supporting looking for a message separator.
Also renaming classes, and removing multi-part functionality is a non-starter, as I'm not up for making any breaking changes at this point, or removing required functionality.

closed 2019-02-08 by dotMorten

hansmbakker, 2019-02-08:
@dotMorten thank you for your feedback! I didn't mean to suggest that I expect you to support this or spend time on this. I understand you don't want to rewrite your own NMEA parser.

My questions were rather whether you would see value in such a generalized library in general and whether you would have any feedback on its usefulness. Maybe I'm trying to look for something that makes no sense, or maybe you heard of a similar generic library that was developed after you started building your NMEA library for example?

dotMorten, 2019-02-08:
@hansmbakker I think that if you have 2 or preferably more scenarios you want to solve, that approach makes sense.
Generally I don't like making abstractions just for the sake of it, unless I know several real-world use-cases that can help me come up with the correct API design to accommodate them all.

hansmbakker, 2019-02-08:
Yes, that makes sense. I also didn't want to create an abstraction layer for the sake of it, but I saw usecase (NMEA parsing) and wanted to do something with interfacing with Grbl (firmware for CNC machines), and could imagine other people interface with other devices.

So I was surprised that there was nothing high-level I could use as a basis for the interfacing + message separation part and focus on the message content parsing rather building than something with System.IO.Ports and maybe file access for testing purposes. I don't expect others to write it, I was just surprised it wasn't there yet besides your library.