dotMorten/NmeaParser #51: UWP Application Using NMEA Parser

jsmcneal, 2019-03-05:
Hello, I am trying to  implement NMEA Parser in a C# UWP application.  I have version 1.9.9.0, but Visual Studio is throwing this error: "No overload for 'device_NmeaMessageReceived' matches delegate 'Event Handler <NmeaMessageReceivedEventArgs>.  Your example has this bit of code:        

private void device_NmeaMessageReceived(NmeaParser.NmeaDevice sender, NmeaParser.Nmea.NmeaMessage args)
      {
         // called when a message is received
      }  

Am I missing something because the NmeaParser does not have a method for device_NmeaMessageReceived?  Am I implementeting incorrectly?

Thank you.

dotMorten, 2019-03-05:
I think some code or info is missing for me to properly answer that.
However, lease see the Wiki: https://github.com/dotMorten/NmeaParser/wiki/Using-in-a-Windows-Universal-App-(SerialPort)
And the sample: https://github.com/dotMorten/NmeaParser/tree/master/src/SampleApp.UWP
and see if that doesn't help you.

jsmcneal, 2019-03-06:
NMEA UWP Wiki references this method:
private void device_NmeaMessageReceived(NmeaParser.NmeaDevice sender, NmeaParser.Nmea.NmeaMessage args)

SampleApp MainPage.xaml.cs has this instead:
private void device_NmeaMessageReceived(**object** sender, **NmeaParser.NmeaMessageReceivedEventArgs** args)

Next Issue:
NMEA's MessageReceived SerialPortDevice event does not appear to be firing.  After confirming GPS traffic on the port via PuTTY, I pulled SampleApp, compiled it, and then ran it against my GPS device.  I observed the same behavior where the event did not fire. (I did verify setup parameters, as well).  

MessageReceived event will fire as expected when using the local data file and LoadLocalGpsSimulationData();.

Was NMEAParser/SerialPortDevice tested against Win10 v1809?

dotMorten, 2019-03-09:
Sounds like the wiki needs fixing.
I last ran the test on 1803, but don't see why it would stop working on 1809. Did you ensure the manifest was set? Are your serial port settings correct?

jsmcneal, 2019-03-11:
Port settings are pretty normal: 9600 baud, No Handshake, 8 Databits, No Parity, One Stop Bit.  As a tertiary test I was able to use System.IO.Ports to manually receive NMEA sentences from my device using SerialPort.ReadLine().

I am not absolutely certain, but I think some changes were made to Windows.Devices.SerialCommunication  between v1803 to v1809.  As I am finding through my own software development stream that v1809 has either broken or changed features in Windows 10 when it comes to UWP apps, but I am no expert.

dotMorten, 2019-03-12:
Does the WPF App work for you? Can you write any UWP app that gets data from the Serial Port?
Also when you run the sample app in this repo, please make sure you update it to use the right COM port name.
In any case as this seems to be an issue with UWP and not this library, I'm going to close this here, but feel free to comment if you believe the bug is specific to this library.

closed 2019-03-12 by dotMorten

jsmcneal, 2019-03-12:
I did not try the WPF app because the software app I am developing is required to use UWP and I have been concentrating on what changed between v1803 to v1809.  We do have a workaround, though:  Since SerialPort appears to be unaffected by 1809, we created a .NET Standard 2.0 library that uses System.IO.Ports and that library is referenced into our UWP project.

Thanks for the help.

dotMorten, 2019-03-12:
I get that, but in order to troubleshoot it would be beneficial to figure out what works and what doesn't.

I don't really get what you did with .net standard 2.0, as SerialPort APIs are not part of the standard.

jsmcneal, 2019-03-12:
Our .Net 2.0 library project also has the Microsoft.Windows.Compatibility package because we are adapting some code from other applications.  

When I grab a few minutes I'll do another static test with  Windows.Devices.SerialCommunication and SerialDevice and see what happens...