dotMorten/NmeaParser #53: NmeaMessage.Parse exception with GPGSV package

tinohager, 2019-03-06:
With this package the parse method throw an exception.

    $GPGSV,3,1,12,02,06,225,16,04,,,40,05,65,251,27,07,40,057,43,0*51
    $GPGSV,3,1,12,02,05,225,17,04,,,42,05,64,250,28,07,39,058,44,0*58

dotMorten, 2019-03-06:
What's the message thrown?

tinohager, 2019-03-06:
> System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation. ---> System.FormatException: Input string was not in a correct format.
   at System.Number.ParseDouble(String value, NumberStyles options, NumberFormatInfo numfmt)
   at System.Double.Parse(String s, IFormatProvider provider)
   at NmeaParser.Nmea.SatelliteVehicle..ctor(String[] message, Int32 startIndex)
   at NmeaParser.Nmea.Gsv..ctor(String type, String[] message)
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Object[] arguments, Signature sig, Boolean constructor)
   at System.Reflection.RuntimeConstructorInfo.Invoke(BindingFlags invokeAttr, Binder binder, Object[] parameters, CultureInfo culture)
   at System.Reflection.ConstructorInfo.Invoke(Object[] parameters)
   at NmeaParser.Nmea.NmeaMessage.Parse(String message)
   at SerialPortProgram.Main(String[] args) in C:\Users\TinoHagerSkiline\Documents\Visual Studio 2017\Projects\Test.Gps\Test.Gps\Program.cs:line 26

commit ee7d6ce1 (2019-03-12): Fixes #53
closed 2019-03-12 by dotMorten