dotMorten/NmeaParser #61: Can't Read from Serial Port on UWP

MelbourneDeveloper, 2019-09-01:
I recently bought a GPS unit. It's an ND-100S. It works. I can get a fix on the GPS signal, and I can get the location within the app that comes with the device, and I can also read the Nmea data on .NET Core. This test passes. I.e. it reads data, and the data is not just an empty list.

https://github.com/MelbourneDeveloper/Device.Net/blob/be795b38f713076a5234eb46054ae19f1314ede2/src/Device.Net.UnitTests/IntegrationTests.cs#L19

I started working on a UWP version to read from the serial port, but I've had no success in getting it to work. I tried multiple variations of code, and nothing seems to work. So, I opened this [Stack Overflow thread](https://stackoverflow.com/questions/57719205/uwp-serial-port-access-code-usually-doesnt-work).

I went looking for an Nmea parser, and found this which is great. It was a double bonus that this supports UWP. So, I slightly modified the sample to get it working on my machine, but I'm getting a similar issue with the sample. No data is ever received from the device. The code is reading from the device in a loop, but the ``device_MessageReceived`` method is never hit. I'm sure it's the right COM port. It almost always reads 0:

[image]

Funnily, it actually worked the first time I ran the sample. But, then it never works again - no matter how many times I connect or disconnect the device. It's as though running the sample once temporarily puts the device out of order and it can't recover. Perhaps a handle is not being disposed somewhere , or something. But, if you've got any ideas on how to get this working consistently, that would be great. 

This is my fork. I've only modified it to make compiling and running the UWP sample straight forward on my machine, and to use the Serial Port access:
https://github.com/MelbourneDeveloper/NmeaParser

MelbourneDeveloper, 2019-09-05:
@dotMorten Hi.

dotMorten, 2019-09-05:
> Hi

@MelbourneDeveloper I'm sorry I don't always instantly respond to questions in a repo with free source code that I hacked together in my limited spare time. Please don't do this.

closed 2020-01-17 by dotMorten