dotMorten/NmeaParser #68: Add iOS Bluetooth device

dotMorten, 2020-01-25:
Brings equivalence with Android and UWP

ramtechjoe, 2020-03-02:
HI Morten,

I don't have a lot of time to look at how you have setup the devices in UWP and Android.  I do have an iOS implementation of connecting to a bluetooth device.  In iOS most devices (the ones I am aware of) do not connect through BLE but through the older Bluetooth which uses the EAAccessoryManager.SharedAccessoryManager.

Also something to keep in mind in iOS.  Not only do you need to include the devices in info.plist, but to deploy through the store you have to have the vendor include you app as allowed to connect to the device.  

I attached the classes I have the manage connecting to the device.  I'm sure your busy at Dev Summit but if you have some time can show how this works in the application.

[AccessorySessionController.zip](https://github.com/dotMorten/NmeaParser/files/4277153/AccessorySessionController.zip)

dotMorten, 2020-03-04:
@ramtechjoe Awesome, thank you Joe. I'll take a look at that after devsummit

dotMorten, 2021-01-20:
Fixed in #5d4686c059354c25759b5c41515d35106a175f56

closed 2021-01-20 by dotMorten