dotMorten/NmeaParser #73: NmeaFileDevice does not stop when EOF

sukesh-ak, 2020-07-16:
Thank you for this component.

While using the following syntax to read out a text file content, after it reaches end of file it starts again from beginning.
How do we stop once its EOF?

```c#
  var device = new NmeaFileDevice(filename, 1000);
  device.MessageReceived += Device_MessageReceived;
            
  Console.WriteLine($"\nOpening file {filename} and sending NMEA strings");
  await device.OpenAsync();

```

dotMorten, 2020-07-16:
This is actually by design since it's meant for simulation (real datasources doesn't suddenly stop). I'd consider adding a property for stopping at the end, or at least an event that you can subscribe to and stop.

sukesh-ak, 2020-07-16:
I agree about looping it for usual device cases other than NmeaFileDevice. 
I am using this library to send data to for testing simulation and unable to send just few lines from the file now.

Is this where the code needs to be changed? I took a quick look
https://github.com/dotMorten/NmeaParser/blob/bbc7cb3589ebc06ebbf69d116c0aefc9dee613ba/src/NmeaParser/BufferedStreamDevice.cs#L128

dotMorten, 2020-07-17:
Yeah perhaps just raise an event here, and check if the stream is still open after the event. That allows you to just cancel out in the event.

closed 2020-07-30 by dotMorten

sukesh-ak, 2020-07-30:
Thank you for the changes @dotMorten

sukesh-ak, 2020-07-31:
Can you publish this to NuGet?

dotMorten, 2020-07-31:
2.1 is not quite ready yet. You can build yourself or get the artifacts from the action

sukesh-ak, 2020-07-31:
Tried building but I think because of all the dependencies, it fails.

sukesh-ak, 2020-07-31:
Ok No worries. I edited the project file and got it working for now.

commit 0bc0c28f (2020-07-30): Adds EndOfStreamReached event for buffered streams and better handle streams that can't auto-rewind

Fixes #73