dotMorten/NmeaParser #74: Can we have an option to disable checksum validation?

sukesh-ak, 2020-08-01:
I am using a data set from a 3rd party custom device which provides NMEA in custom format.
Though the lines start with $, it doesn't have the Talker ID. I did, find and replace with a custom Token to fix that.

But now, the data lines does not come with checksum. So the library throws exception.
Is it possible to add a boolean property like DisableChecksum or so? to skip the checking i such scenarios?

dotMorten, 2020-08-01:
To be a bit pedantic that isn't NMEA then. As you said, it's custom :)
Having said that, can you share a bit of the output log and I can take a look what it would take / check how possible it is?

It might come down to you grabbing the source code and removing the checksum bit: https://github.com/dotMorten/NmeaParser/blob/main/src/NmeaParser/Nmea/NmeaMessage.cs#L156-L174
It actually does skip checksum checking if it doesn't find the `*` character that indicates a checksum is coming.

sukesh-ak, 2020-08-01:
Thanks for the response. Here are few sample lines.
Removing the * might be an easy idea I guess since I am using NmeaMessage.Parse.
BufferedStringDevice (from your test sample) comes with extra features which I don't need like EmulatedBaudRate and BurstRate etc.

```
$,0626492,A,12599907,079593202,0.00,B0@00C3FFD2007A,00CEFFE1008C,00CAFFCB006C,00D5FFDF0087,00D9FFD6008F,00BCFFC20073,00CDFFE4007D,00DBFFC500A6,00BCFFE50076,00DFFFB30086*
$,0626493,A,12599907,079593202,0.00,B0@00DEFFC3009D,00C4FFC2006B,00B9FFD6008A,00D5FFB1007E,00ADFFBE006F,00B1FFCB006E,00DDFFB10090,00C8FFD50084,00E0FFC2008D,00BEFFC400A2*
$,0626494,A,12599907,079593202,0.00,B0@00C0FFB5007D,00C9FFAB00AB,00AAFFCC00AE,00B9FFB3007C,00C9FFCA0093,00CDFFA700B7,00B6FFCB0080,00D8FFB1009A,00C7FFAA009B,00C0FFAA006E*
$,0626495,A,12599907,079593202,0.00,B0@00CFFFAA0093,00C5FFB50099,00D1FFAC0081,00C6FFB30097,00BAFF9D0096,00C5FFA9008A,009CFFC0007B,00CEFF9C009A,00B5FFA90094,00A0FF9D0075*
$,0626496,A,12599907,079593202,0.00,B0@00D2FF9B007B,00BDFFBF00C1,00B5FFAB0077,00A9FFCC0089,00C5FFA80093,00A2FFE10070,00D5FFB7006C,00CAFFBC0093,00BCFFB60069,00C3FFC10064*
```

dotMorten, 2020-08-01:
The spec is pretty clear on what's allowed, and `@` is not a valid NMEA character. There's other validation I just added that would throw when it sees that.

> BufferedStringDevice (from your test sample) comes with extra features which I don't need like EmulatedBaudRate and BurstRate etc.

Because it's meant to be used for emulating real devices. If you just need to rip through data, NmeaMessage.Parse is better.

sukesh-ak, 2020-08-01:
I had written a parser for this data 6yrs back. But in future the data could come from this custom device (have lot of real data in this format for my application testing) or NMEA, I thought its better to use your code and add this custom format for supporting it.

> The spec is pretty clear on what's allowed, and @ is not a valid NMEA character. There's other validation I just added that would throw when it sees that.

Honestly, I don't know what those values are from 'B0@00C3FFD2007A' onwards.
I only use fixtime, Active/Void, Lat, Lon & speed

dotMorten, 2020-08-01:
The data validation is pretty important though. I've got several devices that'll transmit both NMEA and binary data and mix two or more different protocols together. The parser can handle this just fine due to the validation, so I'm a little reluctant to remove that

sukesh-ak, 2020-08-01:
I think its better to fix the data and then send it to NmeaMessage.Parse.
1. Add a custom talkerid
2. Calculate and add checksum at the end (or remove the *)
3. Remove @ symbol in the text

Do you see anything else required to conform to NMEA format?

BTW, Do you have plans to support UBX format in future?

dotMorten, 2020-08-01:
That should do it. 
WRT UBX, I actually just got a couple of those (one being the super accurate one arriving hopefully later today). But no I didn't have any such plans at this point, and if I did it would likely be a separate library (the ublox chip is one of the chips that'll mix NMEA and UBX protocols and hence why I improved message validation this past week - IMHO the mix of protocols is a mis-configuration that's used while still evaluating but not something you'll actually use in real deployments)

dotMorten, 2020-08-01:
Btw I'd be open to also skip checksum validation if there is no checksum after the *. That seems just as reasonable as skipping if the * isn't there.

sukesh-ak, 2020-08-01:
> WRT UBX, I actually just got a couple of those (one being the super accurate one arriving hopefully later today). But no I didn't have any such plans at this point, and if I did it would likely be a [...]

Super accurate = using RTK?
I believe you can also customize the NMEA string to send only selected fields in the case of u-Blox chipsets. 
For example, a mix of data fields from RMC & GGA which is very useful specially to get fields which related to same fix date/time together.

sukesh-ak, 2020-08-01:
> Btw I'd be open to also skip checksum validation if there is no checksum after the *. That seems just as reasonable as skipping if the * isn't there.

Good idea 👍

sukesh-ak, 2020-08-01:
> > BufferedStringDevice (from your test sample) comes with extra features which I don't need like EmulatedBaudRate and BurstRate etc.
> 
> Because it's meant to be used for emulating real devices. If you just need to rip through data, NmeaMessage.Parse is better.

I think you have focused more on data collection and plotting on the map.
In the case of using the library to parse and build analysis, its easier to use BufferedStringDevice and then have a way to disable EmulatedBaudRate and BurstRate etc. (which is what my use case is).

dotMorten, 2020-08-01:
I think you'd get better performance as well as also being able to track messages not being parsed correctly doing this:
```cs

            NmeaMessage msg = null;
            foreach(var line in File.ReadAllLines("filename"))
            {
                try
                {
                    msg = NmeaMessage.Parse(line, msg as IMultiSentenceMessage);
                }
                catch
                {
                    //Parse error
                }
            }
```

dotMorten, 2020-08-01:
Perhaps we need a `public IEnumerable<Tuple<NmeaMessage msg, Exception parseError>> NmeaMessage.ParseAll(string nmeaData);` to simplify that usecase?

sukesh-ak, 2020-08-01:
> Perhaps we need a `public IEnumerable<Tuple<NmeaMessage msg, Exception parseError>> NmeaMessage.ParseAll(string nmeaData);` to simplify that usecase?

Ya something like this would be easier with all the benefits of adding custom message type.

sukesh-ak, 2020-08-01:
> I think you'd get better performance as well as also being able to track messages not being parsed correctly doing this:
>
> ```cs
>             NmeaMessage msg = null;
>             foreach(var line in [...]

This is what I am doing right now.

dotMorten, 2020-08-01:
Made a couple changes to allow you to skip checksum, and also ignore checksums if there's no checksum after the `*`. So all you need to do is inject a message ID after $ (ie `if string.StartWith("$,") { /* fix message */ }` ).
See doc here: https://dotmorten.github.io/NmeaParser/api/netstd/NmeaParser.Messages.NmeaMessage.Parse.html#NmeaParser_Messages_NmeaMessage_Parse_System_String_NmeaParser_IMultiSentenceMessage_System_Boolean_

sukesh-ak, 2020-08-01:
> Made a couple changes to allow you to skip checksum, and also ignore checksums if there's no checksum after the `*`. So all you need to do is inject a message ID after $ (ie `if [...]

That was quick. Thank you so much for the change. Will try it soon.

sukesh-ak, 2020-08-01:
> Made a couple changes to allow you to skip checksum, and also ignore checksums if there's no checksum after the `*`. So all you need to do is inject a message ID after $ (ie `if [...]

Tested now and it works well. Thank you :)

closed 2020-08-01 by dotMorten