dotMorten/NmeaParser #75: Question: What is the use of GnssMonitor?

sukesh-ak, 2020-08-02:
What is the use of GnssMonitor?

For example, if I have the following lines will the GnssMonitor consolidation be based on FixTime of RMC/GGA + GSA?

```
$GPGGA,194156.745,1300.054,N,07959.387,E,1,12,1.0,0.0,M,0.0,M,,*6B
$GPGSA,A,3,01,02,03,04,05,06,07,08,09,10,11,12,1.0,1.0,1.0*30
$GPRMC,194156.745,A,1300.054,N,07959.387,E,510.7,217.1,070720,000.0,W*7B
```

How can I use this while using NmeaMessage.Parse route?
Something like this in future?
https://github.com/dotMorten/NmeaParser/issues/74#issuecomment-667564343

dotMorten, 2020-08-02:
Still work in progress but yes it consolidates messages, GxRMC, GxGGA etc, and tries to only raise one event per burst of data.

sukesh-ak, 2020-08-02:
Cool will wait for it to complete.
Is there a way to get notified when you finish and release the nuget package?

For my current testing the existing library is enough but have some use-case little later, for this.

dotMorten, 2020-08-02:
I'm not away if notifications no. You're always welcome to just build it yourself and use that version.

closed 2020-08-02 by dotMorten