dotMorten/NmeaParser #76: (GnssMonitor) OnLocationChanged query of AllMessages clashes with collection update

HG-Dev, 2020-08-11:
I tacked on the GnssMonitor class to the current NuGet build to test a Bluetooth GPS device with the HoloLens, but accessing properties that come from AllMessages eventually clashes with the message update process. Am I approaching the use of OnLocationChanged wrong, or can this be fixed?

```
private async void OnLocationChanged(object sender, EventArgs e)
{
    await Dispatcher.RunAsync(Windows.UI.Core.CoreDispatcherPriority.Normal, () =>
    {
        TxtBlockGps.Text = $"Timestamp: {DateTime.Now.TimeOfDay.ToString()}\n"
            + $"Latitude: {_monitor.Latitude,7:0.00}° (Accuracy: {_monitor.HorizontalError} m)\n"
            + $"Longitude: {_monitor.Longitude,7:0.00}°\n"
            + $"Altitude: {_monitor.Altitude} m (Accuracy: {_monitor.VerticalError} m)\n"
            + $"Satellites in view: {_monitor.SatellitesInView}";
    });
}
```

[image]

dotMorten, 2020-08-25:
Good catch. I can see how that could be a threading issue.

dotMorten, 2020-08-25:
Thank you for trying it out and giving this valuable feedback. I've pushed the fix.

closed 2020-08-25 by dotMorten