dotMorten/NmeaParser #78: GnssMonitor: picking up GN* messages

HG-Dev, 2020-08-31:
I've been trying to figure out why GnssMonitor.OnMessageReceived has been interpreting exactly one real GGA message before going silent, and I think it might be the m_supportGNMessages flag.

Here's an idea of what my log looks like:
```
(OnMessageReceived) Gga receieved (right after else if (message is Gga gga))
(OnLocationUpdate) {lat, lon, alt, satellites}
(NmeaMessageReceived end) Gga message should have been processed
///
(NmeaMessageReceived end) Gga message should have been processed
(The message contained "GGA" but wasn't picked up by OnMessageReceived? Weird)
///
(NmeaMessageReceived end) Gga message should have been processed
///
(NmeaMessageReceived end) Gga message should have been processed
///
(NmeaMessageReceived end) Gga message should have been processed
///
Ignoring message in favor of Gsv (the m_supportGNMessages return)
(NmeaMessageReceived end) Gga message should have been processed
///
Ignoring message in favor of Gsv
(NmeaMessageReceived end) Gga message should have been processed
```

Assuming my device (a Bad Elf GPS Pro+) supports GN* messages, where are they interpreted?

dotMorten, 2020-08-31:
GN is the combined message. Once it's detected that the device uses GN that is what is used for GGA.

HG-Dev, 2020-08-31:
Is this delivered in one string, or compiled over time? I've confirmed that OnLocationUpdate only works if this GN* flag is ignored.

dotMorten, 2020-08-31:
I'm not really following what your issue is. Could you perhaps share a sample NMEA log from your device?

HG-Dev, 2020-08-31:
Okay. I'll be sure to identify where LocationChanged (whoops, not "OnLocationChanged") is firing.

HG-Dev, 2020-08-31:
[badelf_nmea.log](https://github.com/dotMorten/NmeaParser/files/5148985/badelf_nmea.log)
As noted, LocationChanged is called only once. PropertyChanged is called so long as the given message is not GGA or RMC. I don't see any other areas updating coordinates, though...

dotMorten, 2020-08-31:
Is the location actually changing?

HG-Dev, 2020-08-31:
No. Once the first Gga message is parsed for coordinates, the coordinates (GnssMonitor.Latitude / GnssMonitor.Longitude) never change again, and so LocationChanged is never called again.

dotMorten, 2020-08-31:
If your location isn't changing it makes sense that the location changed event doesn't fire then, right?
Anyway I'll take a look at the log a bit later

HG-Dev, 2020-08-31:
Sure, if I was standing stock still with fantastic precision on the part of my GPS receiver. For this log, I walked in a big circle.

dotMorten, 2020-08-31:
Ok you just said the location wasn't changing. Sounds like we misunderstood each other 😂

dotMorten, 2020-08-31:
Looking at the log, your bad elf is weird ;-) It supports GN messages, but it only uses it for GSA, and everything else is GP for location.
Did you configure your device to report messages like this, or is that just how the bad elf receiver works?

dotMorten, 2020-08-31:
@HG-Dev I'm also seeing in the log you shared that almost all the messages have invalid checksums, and thus fail to parse

HG-Dev, 2020-09-01:
Yeah-- I still don't quite understand what GN messages are, but even I can tell they're not in there. The website specifies that it should have "GGA, GSA, GSV, RMC - Talker ID's GP,GL,GN." I guess I'll contact the maker to confirm what's going on.

The Bad Elf device has only two settings for NMEA, "Default (all NMEA sentences)" and "Simplified NMEA (GGA / RMC only)." I wouldn't worry about the checksums; the text between $ and * all check out with [this website](https://nmeachecksum.eqth.net/).

dotMorten, 2020-09-01:
GP*** : GPS Satellites
GL*** : GLONASS Satellites
etc.
GN***: Combination of all satellites (and usually the one you want to use).

Never fully understood why we need location updates for each constellation, but it makes things tricky because we have to detect that this is a multi-band receiver, and we want to prefer the GN messages over the individual measurements, since this gives the most satellites for the solution fix.

dotMorten, 2020-09-01:
It looks like I should probably detect if there's a GN version for each message type instead. I'll submit a fix for that. Thanks for testing this out and providing the feedback

HG-Dev, 2020-09-01:
Very illuminating, thanks! Glad to help.

dotMorten, 2020-09-01:
You're welcome. It seems rather odd that you're getting `GPGGA` messages but not `GNGGA`. Found this on the @badelf website and it makes no sense to me:
[image]
Basically you only get `GGA` from the GPS system, but no `GGA` from GLONASS, or a combined one of them (I'm assuming `GPGGA` probably is the combined value but using the wrong identifier)

closed 2020-09-01 by dotMorten

HG-Dev, 2020-09-02:
Followup: You're right, GPGGA is the combined value. In their support's words, "the Pro+ is built on a dual-frequency GNSS engine from Mediatek, it defaults to the talker identifier for for GPS (GP) for many of the messages regardless of whether the GLONASS signal is contributing to the solution."

dotMorten, 2020-09-02:
Thanks for the follow-up. That's definitely against the NMEA spec but I can see why. Older NMEA libraries would likely only pass `GP***` messages (including some older versions of this library).
Anyway the latest fix I made should address this

PR #79 (2020-09-01): Detect GN support per-message instead
- Detect GN support per-message instead. Fixes #78
- Simplify Android device lookup