dotMorten/NmeaParser #80: Can't stub messages

stealthrabbi, 2020-09-22:
I'm attempting to create stubs/mocks of Nmea messages, e.g. Gga, using Rhino Mocks, but am getting the error as describe dhere. I'm on .NET Framework 4.7.2.

I think this is related to the libraries use of nullables on the Event definition for MessageReceived. I'm not sure how to get around this.

dotMorten, 2020-09-22:
I have no experience with RhinoMocks, so could you please provide some more information around what you're doing, how you're doing it, what you expect to work and what error are you getting?

In any case, mocking hasn't been a focus here - my main issue with mocking is it usually requires you to increase the API surface beyond what is needed in real use, making it a harder to support and understand API.
You can always create messages using `NmeaMessage.Parse("$GPGGA,,,,,,,,,*00"` if you need to create some on the fly. See the unit tests which are doing that.

dotMorten, 2020-11-27:
Stale. Closing. Please reopen if you have more details to share.

closed 2020-11-27 by dotMorten