dotMorten/NmeaParser #83: High CPU usage

ondrasvoboda, 2020-12-11:
Hi,
I have the following setup:
- WPF app with SharpGIS.NmeaParser 2.1.0
- GPS sensor NL-602U connected via USB

I've noticed high CPU usage when reading data using SerialPortDevice (almost 30%). I have the same results also in your sample SampleApp.NetCore. Is this "normal"? This highly decreases battery lifetime, couldn't it be a bug?

When using sensor native app "u-center" to read the data CPU is only about 1%.

Thanks

dotMorten, 2020-12-11:
The sample app does quite a lot of other things as well like rendering both a 2D and 3D map. The parsing shouldn't be a big contributor to cpu usage.
Have you tried a simple app that does nothing but parsing? What kind of system are you running on?

ondrasvoboda, 2020-12-11:
I've tried following little console app:
```
	class Program
	{
		static void Main(string[] args)
		{
			try
			{
				using (var externalPort = new SerialPort("COM4", 9600))
				using (var externalDevice = new SerialPortDevice(externalPort))
				{
					externalPort.Open();
					externalDevice.MessageReceived += externalDevice_MessageReceived;
					externalDevice.OpenAsync();

					while (true) { }
				}
			}
			catch (System.Exception ex)
			{

			}
		}

		private static void externalDevice_MessageReceived(object sender, NmeaMessageReceivedEventArgs e)
		{
			System.Console.WriteLine(e.Message.ToString());
		}
	}
```
CPU usage is between 40-50%. There is no change if I don't listen to MessageReceived event.
Tested on two computers with Windows 10, using .NET 4.8.

dotMorten, 2020-12-11:
That's expected. Your while loop runs at full speed.
Replace it with Console.ReadLine();

ondrasvoboda, 2020-12-11:
You are right with the while loop I was wondering that it is even higher than in original app. 
When replaced with Console.ReadLine(); CPU usage is little bellow 30% - same as in the original app.

dotMorten, 2020-12-11:
Just tried it. On mine that change made it go from 5% CPU for the process (I have a LOT of processors) to 0-0.1%

dotMorten, 2020-12-11:
That's really odd. The string parsing of NMEA shouldn't take up much CPU load at all. I'm wondering if there's an issue with the USB driver for that device.

ondrasvoboda, 2020-12-11:
Thats a good idea, I'll have a look on the driver. Thanks.

dotMorten, 2020-12-11:
If you can, it would be great if you could pull the source code and run the sample with the visual studio performance analyzers on.
Debug -> Performance Profilers

[image]

I'd be curious to see where the CPU load is happening.

Also are you seeing this happening with the NMEA File-based simulation that the sample app uses on startup?

dotMorten, 2020-12-11:
I see something like this with a console app, and nothing really seem to matter:
[image]
[image]
[image]
[image]

I'm wondering if yours would be significantly different

ondrasvoboda, 2020-12-14:
I've modified console app to just read the data from file, now it look like this:
```
class Program
{
	static void Main(string[] args)
	{
		try
		{
			var fileDevice = new NmeaFileDevice("D://NmeaSampleData.txt");
			fileDevice.MessageReceived += externalDevice_MessageReceived;
			_ = fileDevice.OpenAsync();

			Console.ReadLine();
		}
		catch (Exception ex)
		{

		}
	}

	private static void externalDevice_MessageReceived(object sender, NmeaMessageReceivedEventArgs e)
	{
		Console.WriteLine(e.Message.ToString());
	}
}
```
But in task manager I see CPU around 40%, the same is also in the profiler:
[image]
[image]
It looks like some while loop running, like I had in my first example here.

ondrasvoboda, 2020-12-14:
I believe that the problem is in the while loop in NmeaDevice.StartParser method. When I replace Task.Yield() with 
e.g. Task.Delay(100); CPU usage drops to 1%. But I don't know if there is not any other consequence....

ondrasvoboda, 2020-12-14:
That brings me to StartParser method:
[image]

dotMorten, 2020-12-14:
Thanks. That's curious. The ReadAsync call is meant to be blocking the thread until there's data to read - that however shouldn't cause a lot of CPU cycles at all (so the analysis might show a lot of time spent here, but it's mostly just spent waiting for the device to return some bytes).
I'm curious though if you could add an `else` before the `await Task.Yield();` so it'll only yield if no data was read. I'm wondering if your device/drive just behaves a little bit different here.

ondrasvoboda, 2020-12-15:
Else you suggested doesn't have any effect.
Seems that ReadAsync doesn't wait for the data at all, it very often returns 0 and that is the problem, I think. Do you see a lot of zeroes there too?
If I replace `Task.Yield();` with 
```
else
{
   await Task.Delay(10);
}
```
All is cool and CPU is under 1%.
I'm working with .NET 4.8, maybe it behaves differently in other .NET versions. What are you using?

ondrasvoboda, 2020-12-18:
Another observation - I've tested ppm 10xx GPS sensor and CPU is OK with your latest released version. In this case I don't see any zeroes being returned by ReadAsync. So it might be driver connected, but still I see the same with NmeaFileDevice.

Have you been able to reproduce it with NmeaFileDevice?

One way how to fix this could be to create my own SerialPortDevice, which overrides ReadAsync with Task.Delay mentioned above.

ondrasvoboda, 2020-12-18:
I think I've found it. With my NL-602U GPS there is a lot of exceptions being swallowed here:
```
try
{
	readCount = await ReadAsync(buffer, 0, 1024, token).ConfigureAwait(false);
}
-> catch { }
The I/O operation has been aborted because of either a thread exit or an application request
```
and thats why the while loop is running on a full speed.

I can handle it by customizing SerialPortDevice described above, but maybe you could also fix it in your code.

dotMorten, 2020-12-18:
Good find. What do you mean by "customizing SerialPortDevice" ?
That line of code shouldn't throw.
Would be good if you added a type to the catch and could share the stack trace of the exception as well as any inner exception on the error.

ondrasvoboda, 2020-12-18:
It's System.IO.IOException, no inner exceptions,  stack trace:
```
   at System.IO.Ports.SerialStream.EndRead(IAsyncResult asyncResult)
   at System.IO.Stream.<>c.<BeginEndReadAsync>b__43_1(Stream stream, IAsyncResult asyncResult)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncTrimPromise`1.Complete(TInstance thisRef, Func`3 endMethod, IAsyncResult asyncResult, Boolean requiresSynchronization)
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.ConfiguredTaskAwaitable`1.ConfiguredTaskAwaiter.GetResult()
   at NmeaParser.NmeaDevice.<>c__DisplayClass8_0.<<StartParser>b__0>d.MoveNext() in C:\Projects\nmeaparser\src\NmeaParser\NmeaDevice.cs:line 78
```
By "customizing SerialPortDevice" I meant overriding its ReadAsync method and adding some delay if exception has been thrown or nothing was read. Like:
```
class SerialPortGpsDevice : SerialPortDevice
{
	public SerialPortGpsDevice(SerialPort port) : base(port)
	{ }

	protected override async Task<int> ReadAsync(byte[] buffer, int offset, int count, CancellationToken cancellationToken)
	{
		while (!cancellationToken.IsCancellationRequested)
		{
			int readCount = 0;

			try
			{
				readCount = await base.ReadAsync(buffer, offset, count, cancellationToken);
			}
			catch
			{ }

			if (readCount > 0)
			{
				return readCount;
			}
			else
			{
				await Task.Delay(100);
			}
		}

		return 0;
	}
}
```
This works OK with my device.

dotMorten, 2020-12-18:
The delay doesn't solve the problem. It just reduces the frequency the error happens at. Seems like a driver issue. That line is supposed to sit and wait until bytes are available so adding a delay is problematic.

A delay of 100ms is quite big. You should at least increase the buffer size or you risk running behind and not being able to keep up with incoming data. Perhaps also move the delay into the catch block.

dotMorten, 2020-12-18:
This seems to cover the problem you're experiencing: http://zachsaw.blogspot.com/2010/07/net-serialport-woes.html

ondrasvoboda, 2020-12-18:
Thanks, I'll have a look on it.

ondrasvoboda, 2021-01-06:
It looked promising but it unfortunately didn't work with my device.
I'm closing this issue because the problem is not in NmeaParser.

closed 2021-01-06 by ondrasvoboda