dotMorten/NmeaParser #86: Latitude and Longitude are not update

jankmenta, 2021-02-09:
Hi, I'm using newest version of this Nuget package (2.1.0) in my Android application written in Xamarin and I found that when I received "LocationChanged" event, this event is not fired again. Never. Thus Latitude and Longitude properties in monitor are not updated and they stay in init location. It seems it is similiar bug: https://github.com/dotMorten/NmeaParser/issues/78 but I was hoping it is fixed.

dotMorten, 2021-02-09:
Can you share the code you're using as well as an NMEA log from your device so I can try and repro?

jankmenta, 2021-02-10:
Here is my implementation. It is Xamarin forms application. NmeaParsed is used in Android app.

This code is in Android project in MainActivity.cs:

```
    public class MainActivity : global::Xamarin.Forms.Platform.Android.FormsAppCompatActivity
    {
        protected override void OnCreate(Bundle savedInstanceState)
        {
            base.OnCreate(savedInstanceState);

            Xamarin.Essentials.Platform.Init(this, savedInstanceState);
            global::Xamarin.Forms.Forms.Init(this, savedInstanceState);
            global::Xamarin.Forms.FormsMaterial.Init(this, savedInstanceState);

            LoadApplication(new App());

            // NMEA (GPS)
            var nmeaListener = RootWorkItem.Services.Get<NmeaParser.SystemNmeaDevice>();
            if (nmeaListener == null)
            {
                nmeaListener = new NmeaParser.SystemNmeaDevice(ApplicationContext);
                RootWorkItem.Services.Add<NmeaParser.NmeaDevice>(nmeaListener);
            }
        }
    }
```

And in shared project this is my simplified implementation:

```
    public class GpsService
    {
        private NmeaParser.Gnss.GnssMonitor gpsMonitor;

        public void Start()
        {

            var isRunning = gpsMonitor?.Device?.IsOpen == true ? true : false;
            if (!isRunning )
            {
                if (gpsMonitor == null)
                {
                    // get the platform-specific listener
                    var gpsListener = RootWorkItem.Services.Get<NmeaParser.NmeaDevice>();
                    gpsMonitor = new NmeaParser.Gnss.GnssMonitor(gpsListener);

                    // events
                    gpsMonitor.LocationChanged += GpsMonitor_LocationChanged;
                }

                gpsMonitor.Device.OpenAsync();
            }
        }

        public void Stop()
        {
            if (gpsMonitor?.Device?.IsOpen == true)
            {
                gpsMonitor.Device.CloseAsync();
            }
        }

        private void GpsMonitor_LocationChanged(object sender, EventArgs e)
        {
            // this event is fired only once when location is found.
            // Moreover gpsMonitor.Latitude and gpsMonitor.Longitude is not updated. Does not matter where you try read id
            var lat = gpsMonitor.Latitude;
            var lon = gpsMonitor.Longitude;
        }
    }
```

I've added two log files. Log1 is from NmeaParser (gpsMonitor.AllMessages) Log2 is from external application.

[log1.txt](https://github.com/dotMorten/NmeaParser/files/5957798/log1.txt)
[log2.txt](https://github.com/dotMorten/NmeaParser/files/5957800/log2.txt)

jankmenta, 2021-02-11:
Please ignore log1.txt. I've misunderstood AllMessages property. log3.txt which I've just added should contain all NMEA messages received by NmeaParser library.

[log3.txt](https://github.com/dotMorten/NmeaParser/files/5965496/log3.txt)

dotMorten, 2021-02-16:
I tried replaying your log, but wasn't able to reproduce it.
Note that Android's `SystemNmeaDevice` might not actually produce new locations if you're testing inside.
Wrt the fix in 78, that fix isn't in the latest nuget. Are you building from source or using nuget?

dotMorten, 2021-02-16:
Try v2.2: https://www.nuget.org/packages/SharpGIS.NmeaParser/2.2.0

jankmenta, 2021-02-17:
Thank you for link. I had version 2.1 from nuget. After update to version 2.2 is everythink ok.

closed 2021-02-17 by dotMorten