dotMorten/NmeaParser #91: Trigger callback on read exceptions

denravonska, 2021-04-13:
Maybe I'm approaching this the wrong way, but I'm trying to stream NMEA data from a network device. For this I'm using a TcpClient whose `GetStream()` I feed to `NmeaParser.StreamDevice`. My problem is that since the StreamDevice is doing all the reading I have no way of knowing if the connection is dropped.

I think I can enable TCP keepalive and poll the TCP client for connectivity, but it would be neat if the library can propagate exceptions when reading data so we can have a bit more graceful error handling. Unless there's a better way to do this.

dotMorten, 2021-04-13:
For this scenario what I've typically done is create a new custom Stream class that has your stream inside it and forwards the read calls but also monitors bytes getting through and after a set timeout restarts the stream.

denravonska, 2021-04-14:
@dotMorten That makes sense. Thanks!

closed 2021-04-14 by dotMorten