dotMorten/NmeaParser #97: ReadAsync return just 1 caracter "$"

dylancollaud, 2021-12-02:
Hi, 
I start to use your library in my projet and I've some trouble with it. 
I download and compile it directly on my computer to see what is going on insinde.

The function in the file NmeaDevice.cs line 80
`ReadAsync(buffer, 0, 1024, token).ConfigureAwait(false);`

return juste one char of the stream USB. The char is "$" and flush the line.

I print what your library give me on the event "OnNmeaMessageReceived"

`Message at 2021-12-02 21-14-24$GNGSA,A,3,79,87,88,81,78,,,,,,,,1.58,0.90,1.30*1C`
`Message at 2021-12-02 21-14-27$$$GNGSA,A,3,31,02,25,18,32,26,,,,,,,1.58,0.90,1.30*10`
`Message at 2021-12-02 21-14-34$$$$$GPGSV,3,2,10,20,07,077,,25,64,084,28,26,19,289,30,29,88,185,20*76`
`Message at 2021-12-02 21-14-34$GNGLL,4631.05229,N,00638.72650,E,201432.00,A,A*76`

On the data displayed, I see many "$" without data after.
It's like the stream don't receive all data and I don't know why. Did you already get this problem ?

If I print data directly from USB with windows serial port I receive all data correctly.
I use the ublox M8P-2-11.

I hope you can give me some advice to find the problem. Thank you in advance.

dotMorten, 2021-12-02:
ReadAsync is a partial read. It might not read the entire message, but reads chunks. If only the first character is available, it would just be `$`.
However the next read should read the remaining characters. If that is not happening, make sure your baudrate and all other serial port settings are correct.

dylancollaud, 2021-12-02:
Yes I see this is just a partial read. But with breakpoint each event return "$". 
But my bad... I was sur I changed the baudrate to 115200 and apparently not.. the baudate is still 9600. 
I never saw a device with differente baudrate give some time good value and sentence in this case.. This is why I put fault on the ReadAsync.. 
Sorry for this noob question and thank you for the quick response =)
I'm starting a DIY MowerRTK based on a Husqvarna so I will use and test all of your library =)

closed 2021-12-02 by dylancollaud