dotMorten/NmeaParser #98: CPU usage is high

mclifford1, 2021-12-12:
Hello, 

I've recently updated from using the 1.X to the 2.X version of this (via NuGet) and noticed that the CPU usage of my application is around %40 when running with the debugger, or up to %90+ when running standalone.   Obviously this isn't ideal.

I've isolated the cause as this library and more specifically, the read loop in NmeaDevice (lines 74-92) as it sits in a tight loop.  I've added a delay when no bytes are read as a temp fix.

[image]

Mark

dotMorten, 2021-12-12:
See this comment and the thread below: https://github.com/dotMorten/NmeaParser/issues/83#issuecomment-748048768

You shouldn’t have to add the delay. The readasync method is supposed to do that and not return until data is received.

mclifford1, 2021-12-12:
I have just read through that other bug report, which is very similar to what I am seeing.

For now I am testing with just an NMEA file sample and the behaviour (as with #83 ) is the same in the sample application provided.
[image]

It does seem that the Stream.ReadAsync() returns immediately with 0 bytes most of the time.  I added some debug to capture how many reads it's doing per second:

[image]

I think because BufferedStream is implementing Read in a non-blocking way, the calling ReadAsync is also non-blocking?

If I tweak the Read method to "block" then things calm down a lot:
[image]

[image]

[image]

dotMorten, 2021-12-16:
@mclifford1 Nice catch. Yeah I think the Read should block until the buffer has been populated, but I'd probably do that a bit different.
At least you shouldn't see this high CPU usage with a normal device.

dotMorten, 2021-12-22:
@mclifford1 The above referenced commit brought my CPU usage from ~16% down to between 0 and 0.1%. Thanks for reporting and finding this!

closed 2021-12-22 by dotMorten

mclifford1, 2021-12-22:
Hey no problem, keep up the good work!

commit 6681b8e9 (2021-12-22): Reduce CPU usage. Fixes #98