dotMorten/NmeaParser #99: System.Runtime, Version=4.2.1.0

kastelmedia, 2021-12-15:
Hi,

I am developing a Hololens App with Unity 3D and try to connect a bluetooth GPS device.
I am using Nuget for Unity and SharpGIS.NmeaParser 2.2.0 with uap10.0.16299-> NmeaParser.dll. 
I imported some example code, treid to build in Unity and get the error message:

Assembly 'NmeaParser' with identity 'NmeaParser, Version=2.2.0.0, Culture=neutral, PublicKeyToken=null' uses 'System.Runtime, Version=4.2.1.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a' which has a higher version than referenced assembly 'System.Runtime' with identity 'System.Runtime, Version=4.1.2.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'

I tried to find the Version System.Runtime 4.2.1.0 and it seems not existing? I found only 4.1.2 and 4.3.0, 4.3.1
Please, could you help to fix this ?

dotMorten, 2021-12-22:
I've had this work with HoloLens v1 a while back (albeit I didn't use Unity in this case). If you're getting this, I'm guessing the way you plug this into Unity isn't right and it is referencing an assembly from the wrong target framework

kastelmedia, 2021-12-23:
Thank you for your answer!

If I only import your uap10.0.16299 dll I get the error:
 error CS0234: The type or namespace name 'BluetoothDevice' does not exist in the namespace 'NmeaParser' (are you missing an assembly reference?)

Unity only works with your net451 dll and there is no "BluetoothDevice" available.
I also need to add #if ENABLE_WINMD_SUPPORT to use Rfcomm.
Do you have any hint for me?
Would it be possible to add the 'BluetoothDevice' to the net451 API?

Here is my code and I started with your example:

```
using System;
using System.Collections;
using System.Collections.Generic;
using System.Threading.Tasks;
using UnityEngine;
using System.Linq;
using System.IO;
using System.Runtime.InteropServices.WindowsRuntime;
using System.Threading;
using System.Collections.ObjectModel;
using TMPro;

using NmeaParser;

#if ENABLE_WINMD_SUPPORT
using Windows.Devices.Bluetooth;
using Windows.Devices.Enumeration;
using Windows.Devices.Bluetooth.Rfcomm;
using Windows.Networking.Sockets;
using Windows.Storage.Streams;
using Windows.Networking.Proximity;
#endif

public class GPS_DUAL_SkyPro : MonoBehaviour
{
    public TMP_Text lon;  

    private Queue<string> messages = new Queue<string>(101);

    private NmeaParser.NmeaDevice listener;
    private NmeaParser.Gnss.GnssMonitor monitor;
    private string longitude = "Not changed yet";
    private void Start()
    {
        lon.text = "GPS to read";
    }

    public void callBuetoothDevice()
    {
        Debug.Log("Call LoadDevices");

        LoadDevices();
       
        lon.text = longitude;
    }

    private async void LoadDevices()
        {
#if ENABLE_WINMD_SUPPORT 
        var btdevices = await NmeaParser.BluetoothDevice.GetBluetoothSerialDevicesAsync();
        var XGPS160 = btdevices.Where(t => t.Name.StartsWith("XGPS160-")).FirstOrDefault();
        monitor = new NmeaParser.Gnss.GnssMonitor(XGPS160);
        longitude = monitor.Longitude.ToString();
#endif
        Debug.Log("Call LoadDevices was called: "+ longitude);
}
}
```

dotMorten, 2021-12-23:
Bluetooth device isn’t available in net451. Refer to the API reference on the doc website where it isn’t listed for that target framework

dotMorten, 2022-04-11:
Closing. The pattern you should use is reference based on target framework - you would likely need to mock the bluetooth one when targeting .net framework (that's how I made it work in Unity years ago)

closed 2022-04-11 by dotMorten