dotMorten/NmeaParser #101: SerialPortDevice missing

ondrasvoboda, 2022-01-03:
Hi,
I've noticed that in version 2.2.1 the class SerialPortDevice is missing in .NET Framework build. I think it's because of the bug in constant definition:
```
<PropertyGroup Condition="'$(TargetFramework)' == 'net451'">
    <DefineConstants>$(DefineConstants);NETFX</DefineConstants>
</PropertyGroup>
```
net451 should be net452

Could you please check?

dotMorten, 2022-01-03:
@ondrasvoboda Oops! Thank you. Fix incoming...

closed 2022-01-03 by dotMorten

dotMorten, 2022-01-03:
v2.2.2 published

SonicScholar, 2024-02-26:
> v2.2.2 published

@dotMorten Could you create quickly tag 2.2.2 here?
I spent a little bit of time trying to reconcile the 2.2.2 latest version on NuGet and the release here on your GitHub. Thanks!

dotMorten, 2024-02-26:
@SonicScholar Thanks for the reminder. Looks like I forgot to tag that release. Done!

commit 8c4dbab6 (2022-01-03): Re-adds serial port to NETFX - Fixes #101

Also adds package validation to avoid this happening again