dotMorten/NmeaParser #109: Provide direct connect to NTrip server without mountpoint

ramtechjoe, 2022-12-13:
In some cases a NTrip server might not have multiple mountpoints and connection is done directly with just the IP and Port.  It would be useful if a method was provided for these cases

dotMorten, 2022-12-13:
Can you provide an example for that? Personally never seen this

ramtechjoe, 2022-12-13:
I think it might only be something that is only seen with a private base station.  I am not 100% sure on my terminology, here.  But I think in these cases there is no NTrip Caster, there is an NTrip Source which is connected to directly.

In my case instead of sending a message to an NTrip Caster and getting a list of mount points, then sending the mountpoint name to receive the stream.  A direct connection to the socket is made, and just use that, so this kind of replaces your Request method (although passing in the ip/port)

``` 
    private Socket OpenSocketDirect(string ip, int port)
    {
        Socket socket = new Socket(AddressFamily.InterNetwork, SocketType.Stream, ProtocolType.Tcp);
        socket.Connect(ip, port);
    
        return socket;
    }
```

If there was an OpenStream that had no parameters, this could be used to perform this type of direct connect

```
        public Stream OpenStream()
        {
            if (mountPoint == null)
                throw new ArgumentNullException(nameof(mountPoint));
            if (string.IsNullOrWhiteSpace(mountPoint))
                throw new ArgumentException(nameof(mountPoint));
            
            return new NtripDataStream(() => OpenSocketDirect());  //host and port come from constructor
        }
```