dotMorten/NmeaParser #111: CloseAsync() hangs on an idle NmeaDevice

mclifford1, 2023-04-10:
Hello again,

I have noticed an issue (similar to #88) when calling CloseAsync() on a NmeaDevice (in my case a SerialPortDevice) in that it never returns.  In my case the serial port is idle when this error is observed.

Looking at the source code I believe it is cause by:

[image]

in that the stream does not correctly react to the cancellation token internally (that is, inside the Stream.ReadAsync() itself) and terminate as it should.

This only hangs on an inactive serial port because an active port would be periodically checking the cancellation token and exit read loop appropriately.

I found this thread discussing the issue and possible work-arounds.
https://stackoverflow.com/questions/12421989/networkstream-readasync-with-a-cancellation-token-never-cancels

Best regards,
Mark

dotMorten, 2024-11-20:
See comment here: https://github.com/dotMorten/NmeaParser/issues/113#issuecomment-2487465292
CancellationToken appears to be broken for SerialPort's ReadAsync method.

closed 2024-11-20 by dotMorten

dotMorten, 2024-11-20:
Use this device instead: https://dotmorten.github.io/NmeaParser/api/netwin/NmeaParser.WinRTSerialDevice.html

commit 0be7e552 (2024-11-20): Add WinRT Serial Device

Fixes #111 and #113