dotMorten/NmeaParser #116: .NET 7 causes GPRMC messages to have incorrect FixTime values

justinswork, 2024-09-30:
In .NET 7, the DateTimeOffset `AddSeconds` method was changed for enhanced precision.

The documentation has been updated to explain this new behavior (from https://learn.microsoft.com/en-us/dotnet/api/system.datetime.addseconds?view=net-6.0#remarks)

The issue is explained more in depth in https://github.com/dotnet/runtime/issues/108362

The bug happens here https://github.com/dotMorten/NmeaParser/blob/91d0a73cd29cb1a262532fef25baf1f45d1b64ce/src/NmeaParser/Nmea/Rmc.cs#L50

The end result as far as this package is concerned is that the FixTime of GPRMC messages is "more precise" but it's just wrong.

dotMorten, 2024-09-30:
Note: I haven't had a chance to finish up my .NET 6+ branch just yet: https://github.com/dotMorten/NmeaParser/pull/112

dotMorten, 2024-09-30:
Do you have an example of a string that causes problems?

justinswork, 2024-09-30:
@dotMorten ok, cool. I wasn't aware that branch was in progress. Would be awesome if that could get in soon, so that then .NET 7/8 support could get added as well, this bug doesn't exist in .NET 6 :)

justinswork, 2024-09-30:
> Do you have an example of a string that causes problems?

"$GPRMC,141825.2,A,4249.92297,N,08548.52186,W,000.01,227.1,040322,005.5,W*54"

justinswork, 2024-10-01:
Closing this issue as it is fixed with #117

dotMorten, 2024-10-01:
Reopening since fix isn't merged

dotMorten, 2024-11-20:
Fixed

closed 2024-11-20 by dotMorten